adafruit/Adafruit-PWM-Servo-Driver-Library #13: Documentation is missing

hachi, 2015-04-25:
The product page said the library was documented and had examples. I do see the examples, but I can't seem to find the documentation.

pyropeter, 2015-08-12:
THIS!

justinianpopa, 2015-08-13:
I would also like a nice, formatted documentation with function descriptions. Altough the examples provided have useful line comments for the methods used, not everybody can read and understand those well, especially people new to the C/C++ language so there should be a document to explain each method in a standard fashion.

TomSwift, 2015-09-21:
There's some limited documentation here around page 18:  https://learn.adafruit.com/downloads/pdf/16-channel-pwm-servo-driver.pdf

captainnibbles, 2016-12-02:
This post does a pretty good job of explaining timing, something the DOCUMENTATION should do. 

https://forums.adafruit.com/viewtopic.php?f=50&t=103718#wrap

ladyada, 2018-01-15:
doxygen at http://adafruit.github.io/Adafruit-PWM-Servo-Driver-Library/
and some here https://learn.adafruit.com/16-channel-pwm-servo-driver/library-reference

closed 2018-01-15 by ladyada