adafruit/Adafruit-PWM-Servo-Driver-Library #22: The library does not work with Arduino Due

Hamulus, 2016-09-09:
This library works great with Mega and does not work with Due. It seems like because of some issues in arduino core Wire library.

ladyada, 2018-01-15:
Tested and it definitely...works. just not well :(

closed 2018-01-15 by ladyada

btickell, 2019-11-14:
Is this still an issue? I am trying to run the servo test script and am using identical connections as I have successfully tested with an Uno.

Hamulus, 2019-11-15:
I have not tried new version (the last date when I tested was Sep 9 2016:) ) because I found completely other solution without PCA9685.
However please note, that Due has completely different chip than Uno. In my initial post I mentioned that the library worked perfect with Mega, so Mega and Uno are more similar than Due and Uno or Due and Mega. So if you really want to check, you should use exactly Due, not Uno or Mega.