adafruit/Adafruit-PWM-Servo-Driver-Library #26: setPWM parameters?

EmilyRagan, 2017-03-22:
It is unclear what the parameters of the setPWM method do, mean, or should be, as there is no commenting on that method. The first parameter is the pin number that the servo is connected to, but what are the other two supposed to be? They're called "on" and "off" but what do those numbers mean? What should calling this method actually do?

Thanks.

calston, 2017-07-11:
+1 ... this library is just voodoo

johnyradio, 2018-01-03:
Here ya go:
https://learn.adafruit.com/adafruit-16-channel-pwm-slash-servo-shield?view=all#library-reference

ladyada, 2018-01-15:
hiya see
https://learn.adafruit.com/16-channel-pwm-servo-driver/library-reference
and
https://adafruit.github.io/Adafruit_AMG88xx/html/class_adafruit___a_m_g88xx.html

closed 2018-01-15 by ladyada

calston, 2018-01-23:
Thanks Lady

dantheclam, 2020-11-15:
I  have what is perhaps a related question:  I’m using the setPWM function to drive a servo motor.  Like all the other code I’ve seen online, I let the “on” argument to this function = 0, and let the “off” argument therefore be the parameter that defines the duty cycle of the PWM.    For example when I want to rotate my SG90 micro servo motor (plugged into Channel 0 of a motor driver board), I use this code.    

for(int pos=130;pos<=550;pos=pos+5)
{
pwm.setPWM(0, 0, pos);
delay(100);
}

NOTE that the 2nd argument to the function (the “on” argument) is set to 0.  The numbers 130...550, which I send to the “off” argument span (almost) the full range of motion for this motor.

As an experiment, I tried changing the “on” argument to 700 and adding that same offset to the “off” argument:

for(int pos=830;pos<=1350;pos=pos+5)
{
pwm.setPWM(0,700,pos);
delay(100);
}

Since the duty cycle of the pulse is the same, this works just fine.  But my question is:  why would anybody want to do this?  The SG90 operates at 50Hz, (= period of 20ms), so by setting the “on” position to 700, I suppose that means I’m starting my PWM pulse at  20 * (700/4096) = 3.4 ms instead of at 0ms on each cycle.   To me (as a newbie, doing simple projects) that seems totally unimportant.   After all, the duty cycle is the same and the period between the pulses is the same.   Is it important for some applications that the PWM pulse might began at – say – 3.4 ms in each pulse instead of at 0 ms?