adafruit/Adafruit-PWM-Servo-Driver-Library #34: Compiling for ATtiny85

Shdwdrgn, 2018-02-07:
I am trying to compile the servo.ino example for an ATtiny85 board, however there seems to be no I2C support despite trying various libraries and board cores.

I have tried this on Arduino IDE 1.6.12 and 1.8.5.  I have tried using attiny cores from the following:
https://github.com/SpenceKonde/ATTinyCore
https://github.com/damellis/attiny

I have attempted to compile this while including Wire.h, TinyWireM.h, and TinyWire.h (one at a time of course).  TinyWire.h (available at https://github.com/lucullusTheOnly/TinyWire) seems to be the newest code and is supposed to provide the I2C library via attiny's USI interface.   When I attempt to verify the code, I get the following error message:

    Arduino: 1.6.12 (Linux), Board: "ATtiny25/45/85, Disabled, CPU, ATtiny85, 8 MHz (internal), B.O.D. Disabled"
    
    Build options changed, rebuilding all
    In file included from /home/arduino/projects/attiny/PCA9685_test/PCA9685_test.ino:2:0:
    /home/arduino/arduino-1.6.12/libraries/Adafruit-PWM-Servo-Driver-Library-master/Adafruit_PWMServoDriver.h:53:35: error: expected ')' before '*' token
       Adafruit_PWMServoDriver(TwoWire *I2C, uint8_t addr = 0x40);
                                       ^
    /home/arduino/arduino-1.6.12/libraries/Adafruit-PWM-Servo-Driver-Library-master/Adafruit_PWMServoDriver.h:63:3: error: 'TwoWire' does not name a type
       TwoWire *_i2c;
       ^
    exit status 1
    Error compiling for board ATtiny25/45/85.

The readme for this library states that it is compatible with the attiny85.  Could someone please provide updated directions on what libraries are required together in order to compile the examples for this servo driver board?

ladyada, 2018-02-07:
it works if you use the Wire library we have in our Arduino trinket/gemma board support package

https://github.com/adafruit/Adafruit_Arduino_Boards/tree/master/libraries/Wire

that library implements both tinywire and plain wire at once

closed 2018-02-07 by ladyada

ladyada, 2018-02-07:
you dont want TinyWire, you want just our plain Wire which _integrates TinyWireM,_ it works for our Arduino AVR board kit only! you'll have to do some hacking to add support for other boards. but we don't have a guide or tutorial on how to do so!

Shdwdrgn, 2018-02-07:
Thanks, I finally got it to compile under both the lilypad and gemma boards.