adafruit/Adafruit-PWM-Servo-Driver-Library #40: Change for issue #11 does not address the root cause

thorpej, 2018-05-29:
Hi, I don't think the change for Issue #11 is correct to fix the overshoot problem.  I think the real problem is that the arithmetic used to calculate the prescale doesn't do the rounding step described in the data sheet.  This can lead to the prescale value being too small, which will indeed overshoot the frequency.

Consider setting a frequency of 200Hz (the chip's default).  With the non-rounding arithmetic in the existing driver, you get:

(25000000 / (4096 * 200)) - 1 = 29

...which is NOT the default value of the prescale register.  And it shows the overshoot if you work the math the other direction:

(25000000 / 29) / 4096 = 210

To really fix it, you need to do the rounding step.  This can easily be done by scaling the clock frequency by 100 before calculating the prescale, checking to see if a round-up is required, and then dividing the resulting prescale by 100:

    uint32_t prescale = (25000000 * 100) / (4096 * 200);
    if ((prescale % 100) >= 50)
        prescale += 100;
    prescale = (prescale / 100) - 1;

In the first step, prescale is 3051, in the second step it becomes 3151, and in the last step it becomes (31 - 1), i.e. 30, which is the default value of the prescale register for the documented 200Hz default.

Working that math forward, we can see that the frequency is much closer to what was intended:

(25000000 / 30) / 4096 = 203

I don't think simply setting the frequency to 90% of what the user requested is the right way to solve the problem.

ladyada, 2018-05-30:
ooh - by chance have you tried changing the math in the library, do the frequencies match up right now?

Bolukan, 2018-06-03:
I did a measurement today and suggest to change the oscillator clock frequency in stead of a 10% correction. That is more understandable and probable also more close to the cause of the deviation. For my PCA9685 the oscillator clock is approx. 26076000 MHz. (+4,304%).

Bolukan, 2018-06-08:
I am an irregular programmer and contributor to projects, but added a pull request to realise my suggestion.

Bolukan, 2018-06-09:
Please check, test and comment on this solution: [](https://github.com/Bolukan/Adafruit-PWM-Servo-Driver-Library)

I added an example to calibrate the oscillator clock frequency to be used in the calculations. I also changed the library code to support the use of this calibrated frequency.

If some of you support this direction, I will ask to integrate it in this library.

georgychen, 2018-06-15:
Investigating on the frequency behavior as well. The forward math @thorpej mentioned to estimate update frequency based on a given prescale (ex. 30)  is:
`(25000000 / 30) / 4096 = 203 Hz.`

According to datasheet rev 4 page 25 "The maximum PWM frequency is 1526 Hz if the PRE_SCALE register is set "0x03h". So the math should be
`(25000000 / (3+1)) / 4096 = 1525.8 Hz.`
 
Instead of 
`(25000000 / 3) /4096 = 2034.5 Hz.`

Actually confirmed with a USB oscilloscope that setting the prescale to 0x03h results in 1656 hertz output (~8.5% off). So this formula should get a closer estimate:

`Update Frequency = (OSC_Clock / (prescale +1)) / 4096`

thorpej, 2018-06-15:
@georgychen I think we're essentially in agreement, it's just that there's a boundary condition for deciding when to "+1", and my 200Hz example was one such where it would be appropriate.  My concern would be that "always +1" would sometimes result in a lower update frequency than desired.

Bolukan, 2018-06-15:
If the oscillator clock does not deviate a fixed amount compared to the specs of 25 MHz, but if it depends per item (your 8,5% vs my 4,3%), we should - next to an average adjustment of the standard 25 MHz for non-critical use - make it parameterizable for the users.

I guess the +1 should not be removed as it will not improve the results (predictable frequency)

chrisridd, 2019-10-11:
Separate from the issue of incorrectly multiplying the frequency by 0.9, there is another issue at low frequencies due to the uint8_t cast losing high bits.

If you duplicate the setPWMFreq method but log the prescaleval and prescale values instead of setting them in a register, you can see lots of errors at low frequencies. If you use the correct algorithm from the datasheet, the values look closer to correct. I haven't measured with a scope though!

CSV data:
[PWMFreq.txt](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/files/3718723/PWMFreq.txt)

Graph of 24Hz to 60Hz:
[image]

The "obvious" implementation of the datasheet's algorithm that I used is:

```
void myPWMFreq(float freq) {
  float prescaleval = 25000000;
  prescaleval /= 4096.0;
  prescaleval /= freq;
  int prescale = round(prescaleval);
  prescale -= 1;
  // write to the registers
}
```

Bolukan, 2019-10-11:
@chrisridd, Well found and explained. As the prescale is limited to 8 bits, it should be limited to 255. 
So add one more line (or equivalent):
`if (prescale > 255) prescale = 255;`

chrisridd, 2019-10-11:
That’s another good point @bolukan - In fact the prescale value needs to be constrained between 3 and 255 to handle out of range inputs.

This whole method seems rather flawed. My use case is driving servos, so knowing the exact width of each clock ‘tick’ would be more useful. At least let me set the prescale register explicitly!

thorpej, 2019-10-11:
@chrisridd For the NetBSD driver I wrote for this chip, the pre-scale is set by expressing the period in nanoseconds (consistent with the Device Tree bindings for PWM controllers).  See https://github.com/NetBSD/src/blob/8593ffe98709be14e861cc54566d28b6bd77caaf/sys/dev/i2c/pca9685.c#L355

Bolukan, 2019-10-12:
@chrisridd, hope you can look into the pull request and test yourself.
@thorpej, I chose other formula, but with the same outcome. In the future I would like to change the float to an integer, but as this would change the interface and might hinder users, I left it as it is.

ladyada, 2019-10-14:
we believe this is fixed by @Bolukan 's awesome PR
please try v2.1.0 when it appears in your library manager, and re-open if still an issue!

closed 2019-10-14 by ladyada

chrisridd, 2019-10-28:
I've just had a chance to test the new algorithm in 2.1.0 and can confirm that it works very well and addresses the low frequency issue too.