adafruit/Adafruit-PWM-Servo-Driver-Library #50: Weird bug:  need to explicitly pass in pointer to Wire

sticilface, 2019-07-05:
https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/fa51b6fc8f63b99133b57b3d7b24ac6f9e366bfb/Adafruit_PWMServoDriver.h#L51

I've just spent several hours trying to debug why my project stopped working after months of being fine.  

for whatever reason  (and this applies to the examples as well) 

```cpp
// called this way, it uses the default address 0x40
//Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(0x55);
// you can also call it with a different address you want
//Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(0x41);
// you can also call it with a different address and I2C interface
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(&Wire, 0x55);   // <--------
```

Only the last one worked...  otherwise the arduino would be stuck in a boot loop and never make it past the pwm.begin(). 

I'd like to make a proposal that the initialiser is changed to a reference... my guess is that a hex number can be interpreted as a memory address and it is used as a pointer... 

see this https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/commit/a5068e0fc3cb88e0b5222a8d001e121044cdf693#r34198800

sticilface, 2019-07-05:
Alternatively they could be swapped.  
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(0x55, &Wire);

alesanmanoweb, 2019-07-24:
The change affected a number of older projects, creating an unnecessary incompatibility. Commit 12c492677701109504c430f8edf447f04a9aaced was supposed to be doxygen compliance but instead introduced a break in the API - I'd honestly suggest to reintroduce the Adafruit_PWMServoDriver(uint8_t addr = 0x40); it's only a line that would likely save a lot of grief for less experienced users that find out their perfectly working project is not compiling any longer, making them wary of updating libraries in the future.

ladyada, 2019-07-24:
yep agree
@hoffmannjan did you see this?

hoffmannjan, 2019-07-24:
@ladyada yeah this was unnecessary breaking change, usually we have `i2c` as the last argument. Hope this will fix issues for you guys, if not please reopen.
Thanks for help and sorry for delay!

Bolukan, 2019-10-12:
(Intend but did not succeed in creating separate pull request for this one).
See 4th commit of #61

Bolukan, 2019-10-12:
See #61: last commit contains change for Wire pointer.

ladyada, 2019-10-14:
we believe this is fixed by @Bolukan 's awesome PR
please try v2.1.0 when it appears in your library manager, and re-open if still an issue!

closed 2019-10-14 by ladyada

CheeTay, 2022-10-14:
Hi, my name is KC and very new to this forum. Having similar issue with Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(); as fist mentioned by  #sticilface.
I have reinstalled Arduino IDE 2.0.0 and Adafruit_PWM_Servo_Driver_Library v2.4.1 (latest as of 14/10/2022). The problem I am facing is the code seem to stuck at pwm.begin(); The program is able to print out "Serial.begin..............." in the serial monitor but nothing else. Seems to stuck at code pwm.begin();
my code as follow:
```cpp
#include <Wire.h>
#include <Adafruit_PWMServoDriver.h>

//program able to upload but only able to execute till line, stuck at line 16
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Serial.begin...............");

  pwm.begin();    //execution stuck at this code
  Serial.println("pwm.begin..................");

  pwm.setPWMFreq(50);    //execution stuck at this code
  Serial.println("pwm.setPWMFreq............");
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
Appreciate if someone can help. Apology in advance if my question is not clear, thank you.