adafruit/Adafruit-PWM-Servo-Driver-Library #53: Argument to begin should not set external clock

tbolin, 2019-07-07:
- Arduino board:  Any
- Arduino IDE version (found in Arduino -> About Arduino menu): Any
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):  NA
Calling begin() with an argument will currently set the external prescaler and make the board use an external clock. As far as I know the external clock pin is not accessible on the standard board without modifications. Calling begin with an argument will therefore cause the board to not give any output under most circumstances. This could be confusing to people who are not familiar with the library that might call begin with an argument believing that they are setting the pwm frequency. 

I suggest that the argument to begin is either removed or changed to setting the pwm frequency by calling setPWMFreq.

ladyada, 2019-07-07:
thanks - please submit a PR and we'll take a look!

ladyada, 2019-10-14:
closed due to no respones, reopen if you get back to this :)

closed 2019-10-14 by ladyada