adafruit/Adafruit-PWM-Servo-Driver-Library #54: Method for setting push pull/open drain mode

tbolin, 2019-07-07:
The board is advertised as supporting both open drain and push-pull configurations for the output pins. The library currently does not expose this functionality.

I suggest that a setOutputMode method is added that can change the mode. 
I will add a pull request with the suggested functionality soon. 

- Arduino board:  Any

- Arduino IDE version (found in Arduino -> About Arduino menu):  Any

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): NA

ladyada, 2019-07-07:
thanks - please submit a PR and we'll take a look!

tbolin, 2019-07-09:
Pull request submitted

tbolin, 2019-09-17:
Pull request merged so I'm closing the issue.

closed 2019-09-17 by tbolin