adafruit/Adafruit-PWM-Servo-Driver-Library #58: passing I2C address when creating Adafruit_PWMServoDriver resets Arduino

smarthusker, 2019-07-27:
- Arduino board:  Arduino Uno and Mega 2560

- Arduino IDE version: Arduino IDE 1.8.9 (Windows Store 1.8.21.0)

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
1) Open Arduino IDE using Adafruit-PWM-Servo-Driver-Library v1.0.4.
2) Open example sketch pwmtest.
3) Compile and load pwmtest without any changes to Arduino and verify it runs as expected.
4) Edit line 23 to use ANY I2C address, even default of 0x40.  ie: 
Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(0x40);
5) Compile and load modified example sketch to Arduino.  Arduino will reset.
6) Repeat steps with the example servo.

smarthusker, 2019-07-27:
If Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver(&Wire, 0x40); is used, the code executes as expected.

ladyada, 2019-07-27:
@smarthusker if you use the master branch from github instead of v1.0.4 does that help

smarthusker, 2019-07-28:
I did grab the latest .cpp and .h files that were updated 3days ago without any difference in behavior.

Sent from my iPhone. Please excuse brevity and typos.

hoffmannjan, 2019-08-01:
tested it today with feather PCA9685 and esp32 - all works fine. edited all the examples - PR #59

ladyada, 2019-08-01:
ok merged, @smarthusker plz try master branch again :)

sticilface, 2019-09-28:
Same issue as: https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/issues/50

I still feel that passing in the Wire as a pointer is pointless and unsafe.  using a reference would be much safer and completely prevent errors such as these where the i2c adds is being interpreted as a pointer.

ladyada, 2019-09-28:
@sticilface wanna submit a PR and we'll try it out?

Bolukan, 2019-10-12:
(Intend but did not succeed in creating separate pull request for this one).
See 4th commit of #61
If this the direction, I will adjust the examples too.

ladyada, 2019-10-14:
we believe this is fixed by @Bolukan 's awesome PR
please try v2.1.0 when it appears in your library manager, and re-open if still an issue!

closed 2019-10-14 by ladyada