adafruit/Adafruit-PWM-Servo-Driver-Library #60: Regarding sleep-mode

mrusme, 2019-08-14:
While porting this library onto another platform I noticed some weird things inside the code which I could not explain to myself. Hence I'm writing this issue to clarify how exactly this is intended to work and whether these might indeed be bugs.

This library defines dedicated [sleep](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp#L73) and [wakeup](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp#L83) methods. However, it looks like, during `setPWMFreq` a [redundant implementation](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp#L147) for putting the shield – a **1411** in my case – into sleep mode is being used.

In addition to what seems to be duplicate code, it seems like the code either does not work on Arduino the way it's intended to or there is some other magic happening that allows you to [`setPWMFreq`](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/examples/servo/servo.ino#L46) and thereby put the shield to sleep but then still be able to [directly `setPWM`](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/examples/servo/servo.ino#L71) without explicitly waking it up again. Frankly speaking, I don't understand how this code works on Arduino, but I've tested it myself on an Uno and a Mega2560 and both worked with the [linked example](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/examples/servo/servo.ino).

However, while porting to a different (3.3V operated) platform the shield did not accept any requests for setting the PWM at all while in sleep mode. Only when calling `wakeup` (between `setPWMFreq` and `setPWM`) it would accept the frequency and rotate the servo accordingly.

Maybe someone could explain to me how this code is intended to work and why it does work on Arduino even though the shield is apparently put to sleep right at the beginning? One uneducated guess might be, that the sleep mode itself does not work for this shield while it's being connected to an Arduino, hence there is no need to actually wake it up again.

Thanks in advance!

tbolin, 2019-09-09:
To me it looks like a wake up is sent [two lines after](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/6c1e5b3dd82911271d8ab8dca654bb11e1fd52e1/Adafruit_PWMServoDriver.cpp#L149) it is put to sleep. However, unlike in the sleep method there are no delay directly after sleep mode is set here. I suspect that the board needs some time to go to sleep before the frequency can be set and the board woken up again. If an Arduino Uno or similar an explicit delay might not be needed, but if a faster chip is used weird bugs might pop up. Try adding a delay after [line 147](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/6c1e5b3dd82911271d8ab8dca654bb11e1fd52e1/Adafruit_PWMServoDriver.cpp#L147) and see if it works better.

closed 2019-09-17 by mrusme

tbolin, 2019-09-17:
Did you try it and it worked? If so, the delay should probably be added here too.