adafruit/Adafruit-PWM-Servo-Driver-Library #67: prescale/frequency locked with Library reuse/wrapping of Adafruit 16-channel PWM/Servo

photodude, 2019-11-01:
I'm working with the Adafruit PCA9685 16-Channel Servo Driver and I'm rewriting [this ESC library](https://github.com/RB-ENantel/RC_ESC) to replace the servo library with the [adafruit/Adafruit-PWM-Servo-Driver-Library](https://github.com/adafruit/Adafruit-PW ... er-Library) with this I2C PWM extender. I've tried two different methods (simple object creation like the library I'm rewriting, and full wrapping of all functions) and ended up with the same frequency issue, the I2C PWM extender frequency gets locked to about 24 Hz and will not change. [originally opened in the forums](https://forums.adafruit.com/viewtopic.php?f=22&t=158023) 

- Arduino board: **Uno**

- Arduino IDE version (found in Arduino -> About Arduino menu): **1.8.9**

- Adafruit-PWM-Servo-Driver-Library version: **master branch**

- List the steps to reproduce the problem below (if possible attach a sketch or
copy the sketch code in too): **LIST REPRO STEPS BELOW**

1. Write a `.h` Library header file that creates a class with private object of this library
```cpp
Adafruit_PWMServoDriver servo = Adafruit_PWMServoDriver(0x40);
```
2. Use the servo object in the Library `.cpp` methods
```cpp
void myClass::speed(int speed)
{
writeSpeed = constrain(speed, 1000, 2000);
servo.writeMicroseconds(writeSpeed);
}
```
3. Use this new library in a sketch as per the examples with setup begin set the frequency 60hz, etc
- check the frequency

the expected frequency is 60hz as set
The actual frequency is about 24hz and will not change

I'm at a lost as to why the chip gets frequency locked from creating an object in another library.

Included files are the last attempt at wrapping the library and the associated sketch to use the wapper library
[ESC.zip](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/files/3796762/ESC.zip)

Bolukan, 2019-11-01:
I went through your code, but could not find the obvious fault. Please add 'return's in your wrapper 
```
return myESC.getPWM(num);
return myESC.readPrescale();
```
and use myESC.readPrescale right after ` myESC.setPWMFreq(60)` and `myESC.speed(1064);` and what do you get?

photodude, 2019-11-02:
@Bolukan I made the requested changes (oops forgot those `returns` ::facepalm:: )

I get a prescale of 255

I tried setting the frequency via serial parsing from the serial monitor, I still get a prescale of 255 every time

```
Expected frequency:: 60
Actual prescale:: 255
Expected frequency:: 55
Actual prescale:: 255
Expected frequency:: 2
Actual prescale:: 255
Expected frequency:: 1000
Actual prescale:: 255
```
[I2C_control_Test.zip](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/files/3800115/I2C_control_Test.zip)

photodude, 2019-11-02:
@Bolukan I also tried setting the prescale in the `begin()`, in the sketch with `myESC.begin(105);`
still got 255 for the prescale.

My best guess is something is happening when the private library object is created in header and that something is what's locking the prescale/frequency valuse to 255.

Bolukan, 2019-11-02:
I got your code working, including reading prescale value 106 (instead of 255) with the following three changes:

1. Add `ESC::` to all wrapper functions in cpp: 
```
uint8_t ESC::readPrescale(void)
{
	return myESC.readPrescale();
}
```
2. Only have default values for parameters in the definitions (.h) and not implementation (.cpp):
`		void setPin(uint8_t num, uint16_t val, bool invert = false);`
`	void ESC::setPin(uint8_t num, uint16_t val, bool invert)`

3. Move the initiation of the myESC object to the initalisation of the ESC class:
`Adafruit_PWMServoDriver myESC = Adafruit_PWMServoDriver(oBoard);` ->
`Adafruit_PWMServoDriver myESC;`
to the class initalisation:
```
	oBoard = I2C_Address;	// The I2C address on the TwoWire interface
	myESC = Adafruit_PWMServoDriver(oBoard);
```
And after these changes the code compiled and reported:
```
I2C Controller Test with ESC
255 Current Prescale value
 read Current Prescale value cleared
105 Current Prescale value
```

which is ok, because the first read was before `myESC.begin();`

photodude, 2019-11-02:
Wholely smokes, I didn't even think to separate the library object `Adafruit_PWMServoDriver myESC;` in the header from the object initialization and put that in the CPP `myESC = Adafruit_PWMServoDriver(oBoard);`

I had previously tried moving the whole thing, but not separated. Absolutely brilliant.

It's working as expected with that one minor change (the other changes I had already done, but I guess I had uploaded an old version of the library from earlier version attempt).

I can now change the frequency on the fly and it's working as expected. Thank you. 
(probably should include this solution in some kind of guide)

photodude, 2019-11-14:
I've updated the Forum discussion with the solution from here. Between this and the Forum (and following correct OOP design) anyone else who tries to wrap this library should be able to figure out what to do. 

Closing as solution provided

closed 2019-11-14 by photodude