adafruit/Adafruit-PWM-Servo-Driver-Library #85: Doesn't build

DJSures, 2021-04-30:
I'm not troubleshooting, but instead explaining that your examples do not even build. Is this maintained?

Arduino 1.8.13
ESP32 1.0.6

C:\My Documents\Downloads\Adafruit-PWM-Servo-Driver-Library-master\examples\servo\servo.ino: In function 'void loop()':
servo:103:9: error: 'class Adafruit_PWMServoDriver' has no member named 'writeMicroseconds'
     pwm.writeMicroseconds(servonum, microsec);
         ^
servo:108:9: error: 'class Adafruit_PWMServoDriver' has no member named 'writeMicroseconds'
     pwm.writeMicroseconds(servonum, microsec);
         ^
exit status 1
'class Adafruit_PWMServoDriver' has no member named 'writeMicroseconds'

ladyada, 2021-04-30:
probably something with your setup, you're the only person to have this issue :)

closed 2021-04-30 by ladyada

DJSures, 2021-04-30:
Show me your build success from the example using the library and arduino version specified

ladyada, 2021-04-30:
deleted comment, you can post again, politely

DJSures, 2021-04-30:
C:\My Documents\Downloads\Adafruit-PWM-Servo-Driver-Library-master\examples\servo\servo.ino: In function 'void setup()':
servo:64:7: error: 'class Adafruit_PWMServoDriver' has no member named 'setOscillatorFrequency'
   pwm.setOscillatorFrequency(27000000);
       ^
C:\My Documents\Downloads\Adafruit-PWM-Servo-Driver-Library-master\examples\servo\servo.ino: In function 'void loop()':
servo:103:9: error: 'class Adafruit_PWMServoDriver' has no member named 'writeMicroseconds'
     pwm.writeMicroseconds(servonum, microsec);
         ^
servo:108:9: error: 'class Adafruit_PWMServoDriver' has no member named 'writeMicroseconds'
     pwm.writeMicroseconds(servonum, microsec);
         ^
exit status 1
'class Adafruit_PWMServoDriver' has no member named 'setOscillatorFrequency'

ladyada, 2021-04-30:
hiya, you can browse code here to see that these functions exist
https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp#L341