adafruit/Adafruit-PWM-Servo-Driver-Library #96: Adafruit_PWMServoDriver::setPWM returns 0 not endTransmission response

Carl-UB, 2023-08-10:
Using ESP32 and VSCode with PlatformIO, but this issue doesn't relate to any particular board/IDE.

Adafruit_PWMServoDriver::setPWM documentation states that it returns the result from endTransmission, but in fact always returns 0.

https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/032e425decd24b94333bcf8c3dd322cae37194ce/Adafruit_PWMServoDriver.cpp#L231
vs.
https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/032e425decd24b94333bcf8c3dd322cae37194ce/Adafruit_PWMServoDriver.cpp#L252

It would be good if either the documentation reflected reality, or even better if it actually returned the endTransmission response to allow for detection of errors. This would be similarly useful for Adafruit_PWMServoDriver::writeMicroseconds and Adafruit_PWMServoDriver::setPin.

caternuson, 2023-08-10:
Thanks. Looks like the docstring was not updated when the switch to BusIO was done.

caternuson, 2023-08-10:
OK, should be fixed with #98.

There are some limitations that prevent fully passing the endTransmission return. But it now returns 0 or 1. And docstring updated to reflect that.

Carl-UB, 2023-08-11:
Great, thanks for the speedy fix. Is there a generally preferred choice or rationale between returning a zero code on success vs. a bool indicating success or failure? E.g. Adafruit_PWMServoDriver::begin returns a bool but this returns uint?

caternuson, 2023-08-11:
In that case, just to avoid making a breaking change. The return is currently `uint8_t`. Otherwise would make it a `bool`.