adafruit/Adafruit-PWM-Servo-Driver-Library #105: Resetting of I2C bus speed

retostockli, 2025-04-01:
I make the experience that the command Adafruit_PWMServoDriver::begin() resets the I2C bus speed to 100 kHz. Is this correct? I can set it again to 400 kHz after the begin, but this is not really nice when having a general I2C parameter and bus definition in a larger program context and having many different I2C devices on the bus.

Would it be possible to leave the I2C bus properties untouched of the Adafruit PWM Servo code?

Cheers
Reto

ladyada, 2025-05-26:
yeah i dont think it should, do you want to do a PR?

retostockli, 2025-05-28:
I would like to leave it to the developers to confirm the issue and find the solution if possible. However, If you want I can try to analyze where the problem arises in the library, ok?